carp-dk/carp.sensing-flutter #414: [Carp Serializable][Feature] Allow specifying how to handle serialization errors due to unregistered class type

andreasmpet, 2024-08-29:
Hi! I'm using your carp_serializable library to deserialize json fetched from a Sanity-based backend, and would like for my app to handle new unregistered types gracefully. 

That is, I want that if the backend suddenly returns a json like this:

```
{
"__type": "newUnsupportedTypeFromBackend",
...<other data>
}
```

I could define a closure that returned an arbitrary object if the type is not found. 

To me, this is preferable to the whole json-deserialization failing because the backend suddenly decided to introduce a new type, and I haven't had the time to add support for it yet.

bardram, 2024-09-22:
I've released version 2.0 of the package which should address this issue, using this construct:

```dart
@JsonSerializable(includeIfNull: false)
class B extends A {
  String? str;

  B([super.index, this.str]) : super();

  @override
  String get jsonType => 'dk.carp.$runtimeType';

  @override
  Function get fromJsonFunction => _$BFromJson;
  factory B.fromJson(Map<String, dynamic> json) =>
      FromJsonFactory().fromJson<B>(json, notAvailable: B(-1));
  @override
  Map<String, dynamic> toJson() => _$BToJson(this);
}
```
that specifies a `notAvailable` parameter that is used in the case an uknown type is encountered in the `json`.

closed 2024-09-22 by bardram

andreasmpet, 2024-09-22:
Amazing! Thank you so much for the quick turnaround!

PR #417 (2024-09-22): Serialization v. 2.0 (type safe + graceful error handling)
## 2.0.0

* type safe annotation in class factor method, like this; `FromJsonFactory().fromJson<A>(json)`
* graceful handling of errors when a non-known JSON type is encountered by allowing for a "notAvailable" parameter to the fromJson factory method, like this; `FromJsonFactory().fromJson<B>(json, notAvailable: B(-1))`
* refactor of universal unique IDs (UUIDs) to using the `Uuid().v1` construct
* extending unit test coverage (incl., e.g. exceptions)
* improvement to examples in the `example.dart` file and documentation in the API do [... cut]