carp-dk/carp.sensing-flutter #431: __type not shown in some measurements

sc00n, 2024-10-14:
I updated to   carp_mobile_sensing: ^1.10.0 and here __type was not always added to measurements. For example, I now have measurements like

```
{"sensorStartTime":1728909906404358,"data":{"confidence":100,"type":"STILL"}},
{"sensorStartTime":1728909904458672,"data":{"altitude":51.70000076293945,"accuracy":13.576000213623047,"verticalAccuracy":1.0,"speed":0.0,"speedAccuracy":1.5,"heading":0.0,"headingAccuracy":45.0,"time":"2024-10-14T14:44:54.041","isMock":false,"elapsedRealtimeNanos":924485344000000.0,"elapsedRealtimeUncertaintyNanos":0.0}},
```
while previously, measurments always had a type. e.g.:

`{"sensorStartTime":1726210253753871,"data":{"__type":"dk.cachet.carp.activity","confidence":100,"type":"STILL"}},`

Measurements that currently don't have a type:

- ContextSamplingPackage.CURRENT_LOCATION,
- ContextSamplingPackage.ACTIVITY
- ContextSamplingPackage.WEATHER

Is there a way to add the __type back?

As a side note, most sensors still show a type:
e.g.:

```
{"sensorStartTime":1728909823524384,"data":{"__type":"dk.cachet.carp.screenevent","screenEvent":"SCREEN_ON"}},
{"sensorStartTime":1728909823697560,"data":{"__type":"dk.cachet.carp.batterystate","batteryLevel":52,"batteryStatus":"discharging"}},
{"sensorStartTime":1728909825814826,"data":{"__type":"dk.cachet.carp.screenevent","screenEvent":"SCREEN_UNLOCKED"}},
{"sensorStartTime":1728909839269779,"data":{"__type":"dk.cachet.carp.stepcount","steps":16880}},
{"sensorStartTime":1728909840286456,"data":{"__type":"dk.cachet.carp.stepcount","steps":16881}},
```

bardram, 2024-10-14:
Hm... As you can see, I've been upgrading the serialization to 2.0 (issue #414) and this may have had these consequences. Will take a look.

bardram, 2024-10-14:
> I updated some CARP packages, and now location and activity seems to be fixed. 

Yes - I just tested and the mesurements from `carp_context_package` seems fine.

> I however find problems (no type) in the carp_connectivity_package measures:
> 
> ```
> {"sensorStartTime":1728931956287344,"data":{"connectivityStatus":["wifi"]}},
> {"sensorStartTime":1728932998282828,"data":{"ssid":"\"telenet-A2D179C\"","bssid":"1c:64:99:c0:90:b8","ip":"192.168.0.136"}},
> ```

Will take a look at that package.

bardram, 2024-10-14:
> Will take a look at that package.

Yes - you're right. I didn't update the connectivty package. Has done this now and released as `1.7.1`.

Will check the other packages also.

bardram, 2024-10-14:
Done - as you can see, similar problem in some other packages. Very good that you found this!

closed 2024-10-14 by bardram