carp-dk/carp.sensing-flutter #438: It is impossible to see what device collected a measurement in the data stream

bardram, 2024-10-31:
Looking at the data stream, like:

```json
  {
    "id": 5253,
    "data_stream_id": 639,
    "snapshot": {
      "syncPoint": {
        "synchronizedOn": "1970-01-01T00:00:00Z",
        "relativeClockSpeed": 1.0,
        "sensorTimestampAtSyncPoint": 0
      },
      "triggerIds": [
        10,
        14
      ],
      "measurements": [
        {
          "data": {
            "id": "f2307b9b-e401-4014-b2c7-68e05e2ca17c",
            "__type": "dk.cachet.carp.audio",
            "upload": true,
            "filename": "f2307b9b-e401-4014-b2c7-68e05e2ca17c.mp4",
            "metadata": {},
            "mediaType": "audio",
            "endRecordingTime": "2024-10-29T19:30:46.677579Z",
            "startRecordingTime": "2024-10-29T19:30:34.778037Z"
          },
          "sensorEndTime": 1730230246677579,
          "sensorStartTime": 1730230234778037
        },
``` 

there is no way to know what device collected the measurement or data item(s). 

It is not part of the `Trigger` since in the protocol, this is stored in the `TaskControl`.

How should we store this?

Maybe @Whathecode has thought about this?

Whathecode, 2024-10-31:
That's by design. To interpret the data, you need to look up the ids referenced in the study protocol.

Whathecode, 2024-11-01:
~~That's by design. To interpret the data, you need to look up the ids referenced in the study protocol.~~

Actually, deviceRoleName should be there. The above comment is only true to get the _full_ context (as in, know what type of device that is).

But, the data shown here is much different from e.g. a carp core data point: https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.data.core%2Fsrc%2FcommonMain%2Fkotlin%2Fdk%2Fcachet%2Fcarp%2Fdata%2Fapplication%2FDataStreamPoint.kt

What am I looking at here?

bardram, 2024-11-15:
> Actually, deviceRoleName should be there. The above comment is only true to get the _full_ context (as in, know what type of device that is).

> But, the data shown here is much different from e.g. a carp core data point: https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.data.core%2Fsrc%2FcommonMain%2Fkotlin%2Fdk%2Fcachet%2Fcarp%2Fdata%2Fapplication%2FDataStreamPoint.kt

What is this `DataStreamPoint` used for? As far as I can see, this class is not used for anything in the domain model when I read the Kotlin code @Whathecode ?

> What am I looking at here?

You are looking at a [`DataStreamSequence`](https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.data.core/src/commonMain/kotlin/dk/cachet/carp/data/application/DataStreamSequence.kt) which is part of a list in the [`DataStreamBatch`](https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.data.core/src/commonMain/kotlin/dk/cachet/carp/data/application/DataStreamBatch.kt).

> 
> The "snapshot" looks a bit like a DataStreamSequence. JSON schema: https://github.com/cph-cachet/carp.core-kotlin/blob/develop/rpc%2Fschemas%2Fdata%2FDataStreamSequence.json
> 
> That should have a `dataStreamId`: https://github.com/cph-cachet/carp.core-kotlin/blob/develop/rpc%2Fschemas%2Fdata%2FDataStreamId.json

Yes - this `DataStreamId` seems to be missing in the JSON. 

> Data stream id being an integer is definitely not carp core defined infrastructure, and neither is the "snapshot" property.

Indeed - I will create and issue for CAWS for @yuanchen233  to look at - there is something very wrong here in the data export.

> There are also minor things, like the start and end recording time of audio, which is a clear candidate to use the default timestamps for.

A lot of the data we collect on the phone gets its own timestamp in a UTC format which we save.

> And, as far as I know `__type` needs to be the first defined field for polymorphic deserialization to work.

I hope not - the order of fields in JSON should hopefully not matter - I can't control this in Dart.

bardram, 2024-11-15:
Created an issue i CAWS to address these problems. 

* https://github.com/cph-cachet/carp-webservices-spring/issues/178

closed 2025-07-23 by bardram