carp-dk/carp.sensing-flutter #452: Mismatch between kotlin and dart StudyDeploymentStatusTypes

Panosfunk, 2024-11-22:
[image]

* Need to attach json file here*

bardram, 2024-12-04:
I need the json - not just a picture.

bardram, 2024-12-04:
```
      {
        "__type": "dk.cachet.carp.deployments.application.StudyDeploymentStatus.Running",
        "createdOn": "2024-11-04T20:46:31.356235107Z",
        "studyDeploymentId": "59b39435-e3ef-47cd-b841-fdb842319b98",
        "deviceStatusList": [
          {
            "__type": "dk.cachet.carp.deployments.application.DeviceDeploymentStatus.Deployed",
            "device": {
              "__type": "dk.cachet.carp.common.application.devices.Smartphone",
              "isPrimaryDevice": true,
              "roleName": "Primary Phone"
            }
          },
          {
            "__type": "dk.cachet.carp.deployments.application.DeviceDeploymentStatus.Registered",
            "device": {
              "__type": "dk.cachet.carp.common.application.devices.LocationService",
              "accuracy": "balanced",
              "distance": 10,
              "interval": 60000000,
              "roleName": "Location Service",
              "isOptional": true,
              "defaultSamplingConfiguration": {},
              "notificationOnTapBringToFront": false
            },
            "canBeDeployed": false,
            "remainingDevicesToRegisterToObtainDeployment": [],
            "remainingDevicesToRegisterBeforeDeployment": []
          },
          {
            "__type": "dk.cachet.carp.deployments.application.DeviceDeploymentStatus.Registered",
            "device": {
              "__type": "dk.cachet.carp.common.application.devices.WeatherService",
              "apiKey": "12b6e28582eb9298577c734a31ba9f4f",
              "roleName": "Weather Service",
              "isOptional": true,
              "defaultSamplingConfiguration": {}
            },
            "canBeDeployed": false,
            "remainingDevicesToRegisterToObtainDeployment": [],
            "remainingDevicesToRegisterBeforeDeployment": []
          },
          {
            "__type": "dk.cachet.carp.deployments.application.DeviceDeploymentStatus.Registered",
            "device": {
              "__type": "dk.cachet.carp.common.application.devices.AirQualityService",
              "apiKey": "9e538456b2b85c92647d8b65090e29f957638c77",
              "roleName": "Air Quality Service",
              "isOptional": true,
              "defaultSamplingConfiguration": {}
            },
            "canBeDeployed": false,
            "remainingDevicesToRegisterToObtainDeployment": [],
            "remainingDevicesToRegisterBeforeDeployment": []
          },
          {
            "__type": "dk.cachet.carp.deployments.application.DeviceDeploymentStatus.Unregistered",
            "device": {
              "__type": "dk.cachet.carp.common.application.devices.PolarDevice",
              "roleName": "Polar HR Sensor",
              "isOptional": true,
              "defaultSamplingConfiguration": {}
            },
            "canBeDeployed": false,
            "remainingDevicesToRegisterToObtainDeployment": [
              "Polar HR Sensor"
            ],
            "remainingDevicesToRegisterBeforeDeployment": [
              "Polar HR Sensor"
            ]
          }
        ],
        "participantStatusList": [
          {
            "participantId": "11737f82-3aac-493d-b93b-5cff06e3e4b1",
            "assignedParticipantRoles": {
              "__type": "dk.cachet.carp.common.application.users.AssignedTo.Roles",
              "roleNames": [
                "Participant"
              ]
            },
            "assignedPrimaryDeviceRoleNames": [
              "Primary Phone"
            ]
          }
        ],
        "startedOn": "2024-11-04T20:48:03.458444187Z"
      }
``` 

Also added to the json examples in the app.

closed 2024-12-04 by bardram