carp-dk/carp.sensing-flutter #459: Battery State not updating on Movesense

Goubeast, 2024-12-18:
The battery is always at 80% for the movesense device - see image below

[image]

This may be because  '_addStateSubscription' is not implemented for battery in 'enum MovesenseState' - line 197, carp_movesense_package/lib/movesense_probes.dart

iarata, 2024-12-18:
I think from Movesense we could only get either "OK" (0) or "LOW" (1) for battery status and in the code it was converted to either 80% (OK) or 10% (LOW). See [`carp_movesense_packag/lib/movesense_device_manager.dart`](https://github.com/cph-cachet/carp.sensing-flutter/blob/6903940e2b42e99c74c4dfe26af3af15b8c2106c/packages/carp_movesense_package/lib/movesense_device_manager.dart#L222)

bardram, 2025-02-06:
> I think from Movesense we could only get either "OK" (0) or "LOW" (1) for battery status and in the code it was converted to either 80% (OK) or 10% (LOW). See [...]

Yes - this is true. So, as long as the battery stage is OK, it reads 80%.

We might need to update the UX design to reflect that the battery state can be other than a percentage. @urganerea

closed 2025-02-06 by bardram