carp-dk/carp.sensing-flutter #464: telephony plugin in carp_communication_package is deprecated and no longer maintained

bardram, 2025-02-12:
After upgrading to APK level 18, we get this error from the carp_communication_package:

```
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':telephony'.
> Could not create an instance of type com.android.build.api.variant.impl.LibraryVariantBuilderImpl.
   > Namespace not specified. Specify a namespace in the module's build file. See https://d.android.com/r/tools/upgrade-assistant/set-namespace for information about setting the namespace.
```

This means that the `telephony` isn't upgraded. Moreover, the `telephony` is marked as deprecated and no longer maintained.

bardram, 2025-02-12:
A temporary fix is to manually add the namespace to the `build.gradle` file in the cached version of the plugin.

Edit the `/Users/bardram/.pub-cache/hosted/pub.dev/telephony-0.2.0/android/build.gradle` and add the namespace under the "android" tag, like this:

```
android {
    compileSdkVersion 31
    namespace "com.shounakmulay.telephony"

...

```

Alphaxide, 2025-04-18:
i am stillhaving issues

bardram, 2025-05-23:
Maybe use the [another_telephony](https://pub.dev/packages/another_telephony) plugin?

bardram, 2025-07-23:
Version 1.8 of the plugin uses the "another_telephony" plugin.

closed 2025-07-23 by bardram