carp-dk/carp.sensing-flutter #479: Upgrade to Flutter version 3.29

bardram, 2025-03-13:


Panosfunk, 2025-03-14:
Didn't we say that we can remove carp_esense_package? @bardram

bardram, 2025-03-17:
> Didn't we say that we can remove carp_esense_package? [@bardram](https://github.com/bardram)

Yes. That can be removed from the app - nobody is using these devices in any studies at the moment.

Panosfunk, 2025-03-24:
Check pubspec.yaml files on the packages and make sure that this line 
homepage: https://github.com/cph-cachet/carp.sensing-flutter/tree/main/packages/carp_audio_package
points to the correct branch name (master/main)

Panosfunk, 2025-04-02:
This migration guide can be useful https://docs.flutter.dev/release/breaking-changes/flutter-gradle-plugin-apply

bardram, 2025-04-03:
For the communication package, there is a dependency problem btw. device_calendar and timezone. There is an [issue](https://github.com/builttoroam/device_calendar/issues/586) on this already and I've created a [PR](https://github.com/builttoroam/device_calendar/pull/590).

But we can't publish the communication package before this is fixed - unless we want to point to [my fork](https://github.com/bardram/device_calendar).

PR #484 (2025-04-04): Ugrade cams packages to flutter version 3 29
Closes #479
closed 2025-04-04 by bardram