carp-dk/carp.sensing-flutter #485: Using `health` with `mdsflutter`(Movesense) results on `Undefined symbol: _OBJC_CLASS_$_MDSWrapper` error on iOS

iarata, 2025-04-04:
Currently when using the `mdsflutter` (Movesense) and `health` package together the iOS app will not compile and results in the following error:
```
Could not build the precompiled application for the device.
Error (Xcode): Undefined symbol: _OBJC_CLASS_$_MDSWrapper

Error (Xcode): Linker command failed with exit code 1 (use -v to see invocation)
```

The [Movesense page](https://pub.dev/packages/mdsflutter) suggest to use the dynamic link: [https://bitbucket.org/movesense/movesense-mobile-lib/issues/119/cannot-use-health-and-mdsflutter-depending](https://bitbucket.org/movesense/movesense-mobile-lib/issues/119/cannot-use-health-and-mdsflutter-depending)

Which is to add the following to the `Podfile`
```Ruby
dynamic_framework = ['isar_flutter_libs']

pre_install do |installer|
  installer.pod_targets.each {| pod |
    if dynamic_framework.include?(pod.name)
      Pod::Installer::Xcode::TargetValidator.send(:define_method, :verify_no_static_framework_transitive_dependencies) {}
      puts "turn to dynamic framework name:" + pod.name
      def pod.dynamic_framework;
        true
      end
      def pod.build_type;
        Pod::BuildType.dynamic_framework
      end
    end
  }
end
```

iarata, 2025-04-05:
Also the iOS implementation in `mdsflutter` for the `func connectDevice()` is force unwrapping the UUID causing the app to crash:
```
    public func connectDevice(_ address : String) {
        let uuid = UUID.init(uuidString: address)!
        self.bleController.stopScan();
        self.mds.connectPeripheral(with: uuid);
    }
```

Panosfunk, 2025-04-08:
We had a similar issue with the Studies App. The solution was putting this inside the Podfile

```
target 'Runner' do
  # undocumented flag in cocoapods to enable static linking
  # this is needed so that we can use Movesense and dynamic frameworks together
  use_frameworks! :linkage => :static
  use_modular_headers!

  flutter_install_all_ios_pods File.dirname(File.realpath(__FILE__))
  pod 'Movesense', :git => 'https://bitbucket.org/movesense/movesense-mobile-lib/'
end
```

iarata, 2025-04-08:
@Panosfunk yeah tried that before which was also mentioned by the movesense but still gave the error which eventually was the first comment i put solved it.

PR #487 (2025-04-24): Fixed iOS App
Closes #485
closed 2025-04-24 by bardram