carp-dk/carp.sensing-flutter #488: 'Sensing' and 'one_time_sensing' should be merged into one type called 'sensing'

pavliuc75, 2025-04-24:
From the status meeting today, we agreed that **sensing** and **one_time_sensing** should not be 2 separate types but only one called **sensing.**

bardram, 2025-04-29:
We are getting two `completedapptask` measurement for the `one_time_sensing` now:

```json
{
  "sensorStartTime": 1745924379750743,
  "data": {
    "__type": "dk.cachet.carp.completedapptask",
    "taskName": "Task #20",
    "taskType": "one_time_sensing",
    "completedAt": "2025-04-29T10:59:39.750721Z"
  }
}
{
  "sensorStartTime": 1745924379757487,
  "data": {
    "__type": "dk.cachet.carp.completedapptask",
    "taskName": "Task #20",
    "taskType": "one_time_sensing",
    "completedAt": "2025-04-29T10:59:39.757458Z"
  }
}
```

bardram, 2025-04-29:
> We are getting two `completedapptask` measurement for the `one_time_sensing` now:
> 

That's because the old `OneTimeBackgroundSensingUserTask` should just be removed and always use a `BackgroundSensingUserTask`.

closed 2025-04-29 by bardram